riscv-non-isa/riscv-ap-tee #29: AP-TEE event log

jyao1, 2023-12-06:
Since AP-TEE includes `dynamic_measurements`, should AP-TEE also need `measurement event log`?

Other attestation related spec defined the event log clearly, such as 
 * [TCG Platform Firmware Profile Spec](https://trustedcomputinggroup.org/resource/pc-client-specific-platform-firmware-profile-specification) defined TCG event log
 * [UEFI Spec Confidential Computing Extension](https://uefi.org/specs/UEFI/2.10/38_Confidential_Computing.html) defined (Confidential Computing) CC event log.

Do we want to use similar concept to add RISC-V extension for CC event log?

This is a **direction check**. If we do want to, we can work out next level detail.

For example, we may need add an extension to UEFI spec
```
#define EFI_CC_TYPE_RISCV_COVE  3
```

rsahita, 2024-01-16:
This is a good idea to maintain compatibility @jyao1 - would this add requirements into the CoVE ABI or would it be limited to the UEFI (TVM guest firmware)?

jyao1, 2024-01-21:
My recommendation is:
1) Submit RISCV COVE proposal to UEFI specification. If you agree, I can help on this.
2) Just mention CoVE ABI, that: if a UEFI firmware is used to initialize the RISCV COVE guest environment, then refer to UEFI specification [confidential computing chapter](https://uefi.org/specs/UEFI/2.10/38_Confidential_Computing.html) for CC runtime measurement extension and event log creation.

sameo, 2024-01-21:
> My recommendation is:
> 
>     1. Submit RISCV COVE proposal to UEFI specification. If you agree, I can help on this.

I think we should proceed with that, yes. @rsahita do you agree?
 
> 2. Just mention CoVE ABI, that: if a UEFI firmware is used to initialize the RISCV COVE guest environment, then refer to UEFI specification [confidential computing [...]

Right, so there would not be any CoVE ABI impact, but a only a spec addition describing that if a TVM boots a UEFI firmware, then it may provide the UEFI CC protocol (unless the firmware implements a full virtual TPM iiuc).

jyao1, 2024-01-31:
@rsahita  do we reach consensus to submit UEFI spec change?

rsahita, 2024-02-23:
sorry for the delay - yes lets submit the required request to the UEFI spec

rsahita, 2024-03-10:
@jyao1 @sameo

rsahita, 2024-03-20:
added a note in spec - PR #73

jyao1, 2024-03-23:
Thanks @rsahita, I created 
[CodeFirst - Add RISC-V CC-EventLog 001.docx](https://github.com/riscv-non-isa/riscv-ap-tee/files/14730083/CodeFirst.-.Add.RISC-V.CC-EventLog.001.docx).
Please review it and feedback.

Once we agree the content in AP-TEE TG, we can submit to UEFI together.

jyao1, 2024-03-27:
Thank you @rsahita, I have submitted to https://bugzilla.tianocore.org/show_bug.cgi?id=4738.

Update [CodeFirst.-.Add.RISC-V.CC-EventLog.003.docx](https://github.com/riscv-non-isa/riscv-ap-tee/files/14765993/CodeFirst.-.Add.RISC-V.CC-EventLog.003.docx) with information text - adding URL for AP-TEE.

rsahita, 2024-04-05:
thanks - closing this issue.

closed 2024-04-05 by rsahita