riscv-non-isa/riscv-ap-tee #39: CoVE-IO FID namespace

sameo, 2024-01-05:
The [CoVE-IO](https://github.com/riscv-non-isa/riscv-ap-tee-io) specification extends the CoVE ABI for supporting device direct assignment into TVMs.

The CoVE-IO ABI extensions require their own SBI FIDs and are currently left TBD in the [spec](https://github.com/riscv-non-isa/riscv-ap-tee-io/blob/main/specification/09-coveio_abi.adoc). I propose that we assign a specific FID range for CoVE-IO, so that the CoVE and CoVE-IO FIDs can be formally defined without overlapping or interleaving.

Since FIDs can fill an entire register (`a6`), one possibility would be to reserve the top half `COVG` and `COVH` EID spaces for CoVE extensions. CoVE-IO could e.g take 6 bits from there and use them as needed.

@rsahita @jyao1 WDYT?

jyao1, 2024-01-05:
Yes, I like this proposal to separate role and responsibility.

rsahita, 2024-01-09:
sounds good - we should also make sure there are some reserved portions for other future extensions.

sameo, 2024-01-09:
> sounds good - we should also make sure there are some reserved portions for other future extensions.

Yep.
I'll send a PR for this.

PR #41 (2024-01-20): specification: Define CoVE function ID namespaces
Fixes #39
commit cfc0f6d7 (2024-01-20): specification: Define CoVE function ID namespaces

Fixes #39
closed 2024-01-20 by rsahita