riscv-non-isa/riscv-ap-tee #49: [Qualcomm feedback] Chapter 4: Section 4.2 Clarify TSM must be in confidential memory

rsahita, 2024-02-02:
Reference: [link](https://lists.riscv.org/g/tech-ap-tee/message/128)

> T1: Loss of confidentiality of TVMs and TSM **memory** via in-scope adversaries that may read
TSM/TVM memory via CPU accesses

replace memory with private memory?
There is no definition of "private" memory, I don't believe TSM can use any memory which is not "private". Only TVMs can have both shared and confidential memory.

rsahita, 2024-03-10:
changed this to say:

> T1: Loss of confidentiality of TVMs and TSM confidential memory via in-scope
> adversaries that may read TSM/TVM confidential memory via CPU

rsahita, 2024-03-11:
address in PR #71 

cc @ozkoyuncu

closed 2024-03-11 by rsahita

ozkoyuncu, 2024-03-11:
Agreed,  could not find reference to private memory in the doc either. the comment is a bit confusing as what it is suggesting. I think it is ok