riscv-non-isa/riscv-ap-tee #65: [Qualcomm feedback] Chapter 9.11 onwards (fix ABI errors)

rsahita, 2024-02-06:
Reference: [link](https://lists.riscv.org/g/tech-ap-tee/message/128)

> sbi_covh_add_tvm_page_table_pages

tsm_page_type is not passed as parameter. Missing?

> Copies num_pages pages from non-confidential memory

fix num_pages markup style in 9.11 9.13 and 9.14

> sbi_covh_add_tvm_measured_pages

tvm_guest_gpa missing as potentially invalid address.

> sbi_covh_create_tvm_vcpu

Error code missing for cases where input requirements are not met by tvm_stage_page_addr param.

> 9.16 "0" 

should be SBI_SUCCESS (fix error code in table to SBI_SUCCESS)

rsahita, 2024-02-27:
Fixed in #66 @osman-koyuncu-qualcomm @atishp04

ozkoyuncu, 2024-02-28:
@rsahita  ack. for some reason I don't get the tag notification

eckhard-delfs-qualcomm, 2024-02-28:
@ozkoyuncu : Testing github notification / Please ignore.

rsahita, 2024-03-01:
addressed in #66

closed 2024-03-01 by rsahita