riscv-non-isa/riscv-ap-tee #77: Need MRIF information in sbi_covi_init_tvm_aia() ?

gagachang, 2024-06-25:
In section 7.4.2 TVM external interrupts

```
**tvm_vhart_aia_init**

Initializes the AIA state for a virtual hart. Must be called after the virtual hart has been added but
before the TVM is run for the first time.

The OS/VMM supplies: (1) The guest physical address of the IMSIC for the virtual hart. (2) The
supervisor physical address of a page of confidential memory that is to be used as an MRIF for the
virtual hart. The page is available to be reclaimed upon destruction of the virtual hart. (3) An MSI
address + data pair that is to be signaled when an MSI is delivered to a virtual hart’s MRIF.
```

However, in section 11.1 Function: COVE Interrupt Init TVM AIA (FID #0),
there is no MRIF information in `struct tvm_aia_params`.

atishp04, 2024-07-17:
The current PoC doesn't support MRIF at this point. The above text is bit stale w.r.t implementation. The function names and functionality has been changed . The COVI part describes it correctly. The The absence of MRIF support has been clarified in the above though.

`While it is possible to support execution of a TVM virtual hart using either a
guest interrupt file or an MRIF, the architecture describes below constraints
for the TVM virtual harts to only use guest interrupt files while they are
actively executing in order to simplify the duties of the TSM.`

Are you planning to add support for MRIF to CoVE ? We should clarify the above text for now either way.

gagachang, 2024-07-17:
> The current PoC doesn't support MRIF at this point. The above text is bit stale w.r.t implementation. The function names and functionality has been changed . The COVI part describes it correctly. [...]

Oh OK! If the spec has clarified it somewhere, it should be fine.

> 
> Are you planning to add support for MRIF to CoVE ? We should clarify the above text for now either way.

No~ I just found some inconsistency when I read the CoVE spec, and tried to raise my doubt.
If CoVE's recommendation and PoC use guest interrupt file, that's fine.
Though the spec should cover everything it mentioned before.
Thanks!

atishp04, 2024-07-17:
> Oh OK! If the spec has clarified it somewhere, it should be fine.
> 
> >

> Oh OK! If the spec has clarified it somewhere, it should be fine.
>
>
> No~ I just found some inconsistency when I read the CoVE spec, and tried to raise my doubt. If CoVE's recommendation and PoC use [...]

Yes. The RISC-V IOMMU support patches are being worked in RISC-V community. Once that is matured, this section will be revisited and necessary MRIF support will be added.

gagachang, 2024-07-22:
> Yes. The RISC-V IOMMU support patches are being worked in RISC-V community. Once that is matured, this section will be revisited and necessary MRIF support will be added.

Sounds great! Thanks.

closed 2024-07-22 by gagachang