riscv-non-isa/riscv-ap-tee #85: Use of `enum` is ambiguous

steven-bellock, 2024-08-27:
The specification uses C `enum`s within (packed?) structs. However the size of such fields is ill-defined. This occurs in `AttestationCapabilities` and `MeasurementRegisterDescriptor`.

rsahita, 2025-01-04:
enums should be of size int - also the structs are not packed but what would you suggest? We can use the C23 convention to specify the size of the enum.

steven-bellock, 2025-01-05:
>enums should be of size int

It needs to be able to represent an `int` but the actual storage size is up to the compiler. In particular the RISC-V ABI does not define a fixed size for C `enum`s.

>We can use the C23 convention to specify the size of the enum.

That would work.

>also the structs are not packed but what would you suggest?

Let me file a separate issue for that.

rsahita, 2025-03-11:
@steven-bellock  can you please submit PRs for this one.

steven-bellock, 2025-03-11:
Yes, you can assign it to me.

PR #121 (2025-03-17): Make size of enum in structs explicit
Fix #85.
commit 45b4dcbe (2025-03-17): Make size of enum in structs explicit

Fix #85.
closed 2025-03-17 by rsahita