riscv-non-isa/riscv-ap-tee #119: UEFI reference to RISC-V CC

steven-bellock, 2025-01-03:
#118 inspired me to look at the published UEFI specification, which has https://uefi.org/specs/UEFI/2.11/38_Confidential_Computing.html#risc-v-ap-tee and references AP-TEE, the name of this task group, rather than CoVE. Should this be changed, in a backwards compatible way, to CoVE? Ie

```
#define EFI_CC_TYPE_RISCV_APTEE    3
#define EFI_CC_TYPE_RISCV_COVE (EFI_CC_TYPE_RISCV_APTEE)
```
@jyao1 might know.

rsahita, 2025-01-04:
yeah @jyao1 can you please update? thanks

jyao1, 2025-01-04:
I checked https://github.com/riscv-non-isa/riscv-ap-tee/issues/29.

It seems we use `#define EFI_CC_TYPE_RISCV_APTEE    3` in the ECR.
UEFI 1.11 is just released. Even we submit ECR now. The next release might be another year.

On the other hand, I don't think using TG name is a big issue. E.g, we always use `TCG_EVENT_LOG`, not `TPM _EVENT_LOG`. It just means the TCG defined Event Log. Here it can be interpreted as AP-TEE TG defined CC type.

Please let me know your thought.

rsahita, 2025-01-04:
Seems ok to me to keep it (we can update the note in the COVE spec calling out the  define name used.

steven-bellock, 2025-01-04:
Even in https://github.com/riscv-non-isa/riscv-ap-tee/issues/29#issue-2027465928 you've already got `EFI_CC_TYPE_RISCV_COVE` so that seems like the right choice. Once CoVE is published AP-TEE will probably be forgotten. UEFI can keep `EFI_CC_TYPE_RISCV_APTEE` in UEFI 1.12 for backwards compatibility, but just have it deprecated in favor of the `EFI_CC_TYPE_RISCV_COVE` name.

rsahita, 2025-03-11:
so seems like we don't need to change anything for this one right now.

closed 2025-03-11 by rsahita